leonvandenbeukel/3D-7-Segment-Digital-Clock #5: Pease help to sketch

Billyflora, 2019-05-01:
hi, my name is fabio.
I desperately need your help.
unfortunately I have problems with your sketch and I'm not able to fix it ..
I can kindly and politely ask you if you send me a complete and finished sketch to put only on arduino and send it to billyflora@libero.it?
I did not want to bother you but I have everything ready for 6 months but I am still like a stone.
Thanks a lot, I still hope for your help.
Mondellini fabio

leonvandenbeukel, 2019-05-04:
Hi Fabio,

You can find a complete and finished sketch for Arduino here:

https://github.com/leonvandenbeukel/3D-7-Segment-Digital-Clock/blob/master/3D-7-Segment-Digital-Clock.ino

If you have any issues please be more specific with details of what is going wrong.

Regards,
Leon

closed 2019-05-04 by leonvandenbeukel

Billyflora, 2019-05-04:
hi leon.
I thank you very much for your reply.
I managed to start it, but I would like to put two LEDs per segment instead of one ..
unfortunately I don't know how it gets ...
Would you be so kind as to change yours for my watch project?
I saw yours and I immediately fell in love with it, but with a LED you can't see anything.
I thank you so much again ..
I'm sorry for my goat disability!
I'm trying to learn slowly ...

Billyflora, 2019-05-05:
I'm sorry, another thing ...
I can not understand from the video the lap you did do you have led .. I do not understand the pattern of the turn of the LEDs on the printed numbers ..