leonvandenbeukel/3D-7-Segment-Digital-Clock #6: ciao sono fabio

Billyflora, 2019-05-05:
hi leon.
I thank you very much for your reply.
I managed to start it, but I would like to put two LEDs per segment instead of one ..
unfortunately I don't know how it gets ...
Would you be so kind as to change yours for my watch project?
I saw yours and I immediately fell in love with it, but with a LED you can't see anything.
I thank you so much again ..
I'm sorry for my goat disability!
I'm trying to learn slowly ...

I'm sorry, another thing ...
I can not understand from the video the lap you did do you have led .. I do not understand the pattern of the turn of the LEDs on the printed numbers ..

margaritpoutsi, 2019-11-11:
> hi leon.
> I thank you very much for your reply.
> I managed to start it, but I would like to put two LEDs per segment instead of one ..
> unfortunately I don't know how it gets ...
> Would you be so kind [...]

I would change this bit:
```
void displaySegments(int startindex, int number) {

  **byte** numbers[] = {
    0b00111111, // 0    
    0b00000110, // 1
    0b01011011, // 2
    0b01001111, // 3
    0b01100110, // 4
    0b01101101, // 5
    0b01111101, // 6
    0b00000111, // 7
    0b01111111, // 8
    0b01101111, // 9   
    0b01100011, // º              10
    0b00111001, // C(elcius)      11
    0b01011100, // º lower        12
    0b00000000, // Empty          13
    0b01110001, // F(ahrenheit)   14
  };

  for (int i = 0; i < **7**; i++) {
    LEDs[i + startindex] = ((numbers[number] & 1 << i) == 1 << i) ? (colorMODE == 0 ? colorCRGB : colorCHSV) : colorOFF;
  } 
}
```

to this:
```
void displaySegments(int startindex, int number) {

  **int** numbers[] = {
    0b000111111111111, // 0    
    0b000000000111100, // 1
    0b011001111001111, // 2
    0b011000011111111, // 3
    0b011110000111100, // 4
    0b011110011110011, // 5
    0b011111111110011, // 6
    0b000000000111111, // 7
    0b011111111111111, // 8
    0b011110011111111, // 9   
    0b011110000001111, // º              10
    0b000111111000011, // C(elcius)      11
    0b011001111110000, // º lower        12
    0b000000000000000, // Empty          13
    0b011111100000011, // F(ahrenheit)   14
  };

  for (int i = 0; i < **14**; i++) {
    LEDs[i + startindex] = ((numbers[number] & 1 << i) == 1 << i) ? (colorMODE == 0 ? colorCRGB : colorCHSV) : colorOFF;
  } 
}
```
and this:
  ```
  displaySegments(**0**, mr);    
  displaySegments(**7**, ml);
  displaySegments(**16**, hr);    
  displaySegments(**23**, hl); 
```

to:
```
  displaySegments(**0**, mr);    
  displaySegments(**14**, ml);
  displaySegments(**30**, hr);    
  displaySegments(**44**, hl); 
etc.
```

Dante6214, 2019-11-22:
Hi, the easiest way to make 2 LEDs per segment is to make a diagram so that there is a parallel connection of 2 LEDs as in the picture. And absolutely no need to change the sketch. 
The picture is available at  link:
https://yadi.sk/i/FBnUcPGQ3qxf7w

pakobrien, 2021-04-17:
I do this but my dots don´t blink,the segment number 8 (leds 15 and 16)are always on and blinking,any solution?