leonvandenbeukel/3D-7-Segment-Digital-Clock #22: Problems compiling

m0nde, 2021-05-05:
I am no expert in Arduino and trying to build this clock for my son. I have built the hardware and he has printed all the clock pieces. When I compile the code for the nano I get this error.  Can someone please help me get past this issue please?
Arduino: 1.8.13 (Windows 10), Board: "Arduino Nano, ATmega328P"

In file included from C:\Users\nigel\Documents\Arduino\libraries\src/bitswap.h:4:0,

                 from C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock\3D-7-Segment-Digital-Clock.ino:1:

C:\Users\nigel\Documents\Arduino\libraries\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\nigel\AppData\Local\Temp\ccKsavlE.ltrans0.ltrans.o: In function `displayClock':

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:267: undefined reference to `RTC_DS3231::now()'

C:\Users\nigel\AppData\Local\Temp\ccKsavlE.ltrans0.ltrans.o: In function `processCommand':

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:240: undefined reference to `DateTime::DateTime(unsigned int, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:240: undefined reference to `RTC_DS3231::adjust(DateTime const&)'

C:\Users\nigel\AppData\Local\Temp\ccKsavlE.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_3D_7_Segment_Digital_Clock.ino.cpp.o.4410':

<artificial>:(.text.startup+0x1bc): undefined reference to `Timer::Timer()'

<artificial>:(.text.startup+0x1c4): undefined reference to `Timer::Timer()'

<artificial>:(.text.startup+0x1cc): undefined reference to `Timer::Timer()'

C:\Users\nigel\AppData\Local\Temp\ccKsavlE.ltrans0.ltrans.o: In function `setup':

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:118: undefined reference to `RTC_DS3231::begin()'

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:123: undefined reference to `RTC_DS3231::lostPower()'

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:126: undefined reference to `DateTime::DateTime(__FlashStringHelper const*, __FlashStringHelper const*)'

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:126: undefined reference to `RTC_DS3231::adjust(DateTime const&)'

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:129: undefined reference to `Timer::every(unsigned long, void (*)())'

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:130: undefined reference to `Timer::every(unsigned long, void (*)())'

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:131: undefined reference to `Timer::every(unsigned long, void (*)())'

C:\Users\nigel\AppData\Local\Temp\ccKsavlE.ltrans0.ltrans.o: In function `loop':

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:137: undefined reference to `Timer::update()'

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:138: undefined reference to `Timer::update()'

C:\Users\nigel\Documents\Arduino\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock-master\3D-7-Segment-Digital-Clock/3D-7-Segment-Digital-Clock.ino:139: undefined reference to `Timer::update()'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board Arduino Nano.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Alex-Ender35, 2023-06-15:
Hello, m0nde, you managed to solve your problem with the compilation errors. If so, how?

Alex-Ender35, 2023-06-15:
Какие библиотеки вы использовали? Или может быть внесли изменения в коде прошивки?

m0nde, 2023-06-15:
Sorry no luck here

kind regards

Nigel

Alex-Ender35, 2023-06-15:
> Извините, здесь не повезло, с уважением, Найджел.

Genasaf, 2024-05-09:
Все работает скетч грузится