leonvandenbeukel/3D-7-Segment-Digital-Clock #28: Google play missing app

tamgor, 2024-03-19:
I changed my phone, unfortunately there is no more app on google play.

Im asking for help

ZioRaff, 2024-04-06:
I have the same problem but can't I get the app code?

Genasaf, 2024-04-14:
Поделитесь файлом загрузки приложения приложения

pjaspinski, 2024-07-03:
I made a Python script that can connect to the clock and set current time. You can find it here: https://github.com/pjaspinski/digital-clock-toolkit

Another way of communicating with the clock is to get an app that can send serial Bluetooth 2 messages. I successfully used this app: https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal
If you look into `processMessage` function in clock's code, you can deduct format of messages. For example, message used to set time has following format: `|RTC,2024,07,03,19,55,06`.